derikon/TactJam #1: decide about the license

derikon, 2020-10-19:
Hardware design, PCB schematics, and supplementary materials should be licensed under creative commons. Which type is suitable, e.g. public domain?

derikon, 2020-10-29:
Here are some basic considerations to be taken into account: [https://opensource.guide/legal/](https://opensource.guide/legal/)

What about the combination of multiple licenses in one project - one for code and another for schematics, CAD-files, etc.?

freitaggeorg, 2020-11-10:
Maybe we can ask Lucas here, if he has already seriously studied this topic.

derikon, 2020-11-17:
We can use multiple licenses in a single project. It's good pratice to add a license-file in the root directory that provides further information about the licenses for certain sub-directories.

derikon, 2020-12-21:
I would use [CC-BY](https://creativecommons.org/licenses/by/4.0/)

closed 2020-12-21 by derikon