derikon/TactJam #2: decide on the software license

derikon, 2020-10-19:
Software should be licensed under Open Source. Which one is suitable, e.g. MIT?

derikon, 2020-10-29:
Here are some basic considerations to be taken into account: [https://opensource.guide/legal/](https://opensource.guide/legal/)

What about the combination of multiple licenses in one project - one for code and another for schematics, CAD-files, etc.?

freitaggeorg, 2020-11-10:
The same approach as with the hardware - ask Lucas :)

derikon, 2020-11-17:
We can use multiple licenses in a single project. It's good pratice to add a license-file in the root directory that provides further information about the licenses for certain sub-directories.

derikon, 2020-12-21:
I would use the [MIT license](https://choosealicense.com/licenses/mit/) for our software.

closed 2020-12-21 by derikon