derikon/TactJam #4: documenting placement of actuators

derikon, 2020-11-02:
How do participants document their actuator setup for each tacton? This is important in terms of reproducibility.

Hypnozk, 2020-11-09:
Google doc with the possibilites(pros,cons) https://docs.google.com/document/d/16SL6cMSfG9839DYE07V-BWFXGv0OAmQ8QSwlcOuKfSk/edit?usp=sharing

Hypnozk, 2020-11-10:
decision 4 prepared; because it's easy to handle and high usability

derikon, 2020-11-10:
@Hypnozk, could you edit your last comment and add some more information about number 4? That will help people who don't have access to the document. Thanks

Hypnozk, 2020-11-13:
next steps: decide how to implement the choosen option; it would be possible to show the sketches on a pc software or use maybe other programs there you could work together on the positions

Hypnozk, 2020-11-27:
we decide to prefer to mark the actuators on a sketch in the pc software; but there are some remaining questions on the behaviof functions of the software
next steps: define the functional requests for the software

closed 2020-11-27 by Hypnozk