derikon/TactJam #5: varying intensity of actuators

derikon, 2020-11-02:
## Question

Should we consider varying the actuators' intensity in record mode?

## Ideas

1. enable a potentiometer to vary the intensity while playing live
2. first, record "the beat" and modulate the intensity in a second pass while playing back "the beat" (e.g. in loop with overdubbing) an record the intensity modulation solely
3. use force sensitive resistors (read pressure and map it to intensity levels) instead of push buttons

freitaggeorg, 2020-11-10:
Setting the intensity is, in my opinion, a relevant option for Tacton design. On the other hand, the focus is on the temporal sequence of the individual actuators. For this reason, I find option 2 most useful. Creating the beat and then modulating it in a separate step.

derikon, 2020-11-10:
In the last meeting we talked about this topic. We "decided" to focus on location and time (rhythm) and don't allow a second pass for modulations. But that can still be discussed.

derikon, 2020-11-13:
# We will roll back to the two-pass strategy

1. record rhythm wit a global intensity value
2. [optional] record modulation of the intensity

Intensity values can be changed via a potentiometer or a slider.

closed 2020-11-15 by derikon