derikon/TactJam #8: Do we want to use a visualization for tactons?

derikon, 2020-11-03:
That might be more user-friendly and would give us a way to, first, see if the tactons are recorded correctly (which would also be good for users when they record tactons), and second, report visually the outcomes of the tactile jamming sessions

Here is an [example of a design](https://kingsx.cs.uni-saarland.de/owncloud/index.php/s/tyX4rKL7H5Ke5AX).

bfrucha, 2020-11-09:
To me, the visualization would be great to report the results of the workshop or other focus groups we migth lead using TactJam. Thus, we **need** one.

Now, I also think it's important that users do not see visually tactons while playing with TactJam, because that would bias their creativity (I am relatively certain they will focus on the visual aspect more than the tactile aspect). So we also **need** a way to disable this visualization while the users are designing tactons.

freitaggeorg, 2020-11-10:
Possibly there is only a visual representation as a kind of preview, if the file is stored on the computer. This means that the preview can only be applied to saved tactons and is not visible during composition (recording). 
Otherwise I agree with the points mentioned above.

derikon, 2020-11-13:
# We will implement visualization

closed 2020-11-13 by derikon