derikon/TactJam #15: Recording of actuator position

Hypnozk, 2020-11-09:
We have to decide if we need the information, where the actuator are positioned for researches, etc.

Hypnozk, 2020-11-09:
Google doc with the possibilites(pros,cons) https://docs.google.com/document/d/16SL6cMSfG9839DYE07V-BWFXGv0OAmQ8QSwlcOuKfSk/edit?usp=sharing

derikon, 2020-11-09:
Hi @Hypnozk! Thannks for sharing the document. A issues related to this topic already existed (see issue #4)
Please merge all information into one issue and delete the other.

derikon, 2020-11-09:
> We have to decide if we need the information, where the actuator are positioned for researches, etc.

We have to document this information. The question is how we can do this in an appropriate way.

closed 2020-11-09 by Hypnozk