derikon/TactJam #16: How do we record tactons?

bfrucha, 2020-11-09:
- Do we record tactons using the software side (i.e., TactJam must continuously send information to the software)?
- Do we set a time limit for each tacton? That would make the recording simpler, but that would be very constraining for the users.
- Do we want users to be able to record a tacton without using the software UI at all?

freitaggeorg, 2020-11-10:
In my imagination the frontend has following functions:
- recording of tactons
- select a recorded tacton and
- replay it
- manipulate its intensity 
- add meta information (position of the motors)
- delete it

In generell the frontend enable users to import and export tactons to a predefined interface/backend

derikon, 2020-11-10:
@bfrucha: very important questions!

> Do we record tactons using the software side (i.e., TactJam must continuously send information to the software)?

If I remember correct we decided to record the tacton on the TactJam-device and send it to the PC-software afterwards. I tried to sketch this in one of the sequence diagrams. We can discuss this in more detail on Friday. I will put it on the agenda.

> Do we set a time limit for each tacton? That would make the recording simpler, but that would be very constraining for the users.

Indeed, there is some kind of time limit. The max. recording length depends on the storage size if we buffer the tacton on the micro-controller.

> Do we want users to be able to record a tacton without using the software UI at all?

I would appreciate to be able to do that. To me it seems to be little bit distracting to interact with two devices at the same time. On the other hand we can reduce the device's complexity to just the push button and do the rest in the GUI software (PC). That would fit the functionality that @freitaggeorg described. Let's discuss this on Friday too.

closed 2020-12-21 by derikon