derikon/TactJam #24: Schematic and PCB

M6QF, 2020-11-17:
According to Hardware interface v0.1.
[TactJam_PCB_v0.1.pdf](https://github.com/derikon/TactJam/files/5553816/TactJam_PCB_v0.1.pdf)
[TactJam_Schematic_v0.1.pdf](https://github.com/derikon/TactJam/files/5553817/TactJam_Schematic_v0.1.pdf)

M6QF, 2020-11-17:
Next steps:
- PCB to CAD
- Add buzzer and resistor
- Add low-pass filter and Schmitt trigger for buttons
- Upload EasyEDA file

derikon, 2020-11-20:
# subtasks

@M6QF, following the tasks we discussed.

## misc
1. test daisy chain of 74HC166
2. test LEDs in parallel and in series with actuators

## schematics
1. detach ESP32 3V3 and 5V
2. create semantic groups
3. add buzzer (+resistor)
4. LED for each actuator (SMD)

## PCB (dev board)
1. all parts on top side
2. double row pins for ESP32
3. external power connector (TH micro USB)
4. LED for each actuator (SMD)

M6QF, 2020-11-24:
Updated Dev Board version:

[TactJam_DevBoard_Schematic_v0.1.pdf](https://github.com/derikon/TactJam/files/5589893/TactJam_DevBoard_Schematic_v0.1.pdf)

derikon, 2020-11-26:
Hi @M6QF, I started reviewing the schematic.

### RC filter
voltage levels taken from ESP32 datasheet (5.3 DC Characteristics)
VDD = 3.3V
(high) V_IH = 0.75\*VDD = 2.475V
(low) V_IL = 0.25\*VDD = 0.825V

I used the formula from [https://my.eng.utah.edu/~cs5780/debouncing.pdf](https://my.eng.utah.edu/~cs5780/debouncing.pdf).
t = 20ms
C = 1uF
**R_1 = R_2 = ~14.4k**

Please tell me, how you calculated the resistor values as they differ from mine.

### Piezo Buzzer
Same configuration which I tested on a breadboard. Works 👍 

### Rotary Switches
Here, I don't know what I should review. Looks like you are adding a 1k resistor per step. So the analog output should be 0.0—3.3 volt in 8 steps (each 0.4125V). Is this correct?

### Step-Down Converter
I have no knowledge about this kind of circuit. If you provide more details which part/configuration I should review I can try to read about this topic. Or ask @sidecutter

M6QF, 2020-12-01:
Updated: 
- EasyEDA Schematic file
- EasyEDA PCB file
- BOM
- Pick & Place file
- Gerber

[TactJamDevBoard.zip](https://github.com/derikon/TactJam/files/5619526/TactJamDevBoard.zip)

closed 2020-12-03 by M6QF