derikon/TactJam #45: Find compatible connector (female) for C132463

derikon, 2020-12-03:
The actuators on the dev-board are conntect via XH header [C132463](https://lcsc.com/product-detail/Wire-To-Board-Wire-To-Wire-Connector_Ckmtw-Shenzhen-Cankemeng-C132463_C132463.html). We need to find the compatible counter part (female).

M6QF, 2020-12-21:
Perhaps we can use something like this:
- https://www.digikey.de/product-detail/de/jst-sales-america-inc/XHP-2/455-2266-ND/555485
- https://www.digikey.de/product-detail/de/molex/0351550200/WM3137-ND/2405551
- https://lcsc.com/product-detail/Rectangular-Connectors-Housings_JST-Sales-America_XHP-2_JST-Sales-America-XHP-2_C144401.html

derikon, 2020-12-21:
@M6QF: As far as I can see from the images none of the proposed connectors seem to fit. In case we can change the female connectors if you can find any corresponding parts.

Alternatively, why not just use a 16pin-connector:

- https://lcsc.com/product-detail/New-Arrivals_JST-Sales-America-XHP-16_C495639.html
- https://lcsc.com/product-detail/Wire-To-Board-Wire-To-Wire-Connector_JST-Sales-America-S16B-XH-A-LF-SN_C157933.html

I am not 100% sure if they fit perfectly...

closed 2021-01-14 by derikon