derikon/TactJam #47: How to read remaining RAM

derikon, 2020-12-09:
If users record a very long tacton we may run out of memory on the ESP32. How can we check the remaining RAM size at run time?

lhinderberger, 2020-12-09:
I suggest we don't but instead reserve a reasonably long buffer, limiting the theoretical length of patterns.
The amount of memory that VTP patterns take up depends largely on whether the input changes rapidly or not.

To find out how long that buffer should be, we should think about how many VTP instructions per second we can reasonably expect and what maximum length of patterns would be reasonable. It is worth noting in our context that if a parameter changes for all channels, that only takes up one instruction, instead of one per channel.

Example: If we're looking at a scenario of constantly 25 instructions per second (which is likely much more than what we'll see in practice, if I'm estimating it correctly based on my limited knowledge of TactJam), we'd have 100 bytes per second (a VTP instruction word is 32bits). So for a one minute pattern that would require a 6K buffer. Assuming only 5 instructions per second (which is probably closer to what can be expected), that 6K buffer could hold 5 minutes. And with the available 520K on an ESP32, we could easily go up to 20-30K buffers, assuming there's no other memory intensive applications around :)

lhinderberger, 2020-12-09:
By the way, do we use an OS on our ESP32 or is it bare metal? In other words: Is dynamic memory allocation available?

derikon, 2020-12-09:
> By the way, do we use an OS on our ESP32 or is it bare metal? In other words: Is dynamic memory allocation available?

@lhinderberger: The ESP is running [FreeRTOS](https://freertos.org/) and dynamic memory allocation is available. I quickly tested it (see commit [
72ff265](https://github.com/derikon/TactJam/commit/72ff2657212741ba90f5997f0b03820f55cf1f96)).

derikon, 2020-12-09:
Espressif documentation: [Heap Memory Allocation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html)

closed 2020-12-21 by derikon